Kragrathea/OctoPrint-PrettyGCode #23: Feature request: Show 'Load GCode visualizer' button

Stretch96, 2020-04-06:
When the GCode file is over the 'GCode visualizer file size threshold' setting, PrettyGCode doesn't display anything until the user goes to the 'GCode Viewer' tab, and accepts to load the file

Having the button to load the file within the PrettyGCode tab will prevent users wondering why it's not showing anything

Kragrathea, 2020-04-19:
I didn't know there was a size threshold. I'll look into it.

sr1329, 2020-08-19:
You could change the values in GCode Visualizer settings.

Kragrathea, 2020-08-19:
Increasing the size is a good workaround. But PrettyGCode should do a better job handing the failure. I need to dig into how the built in gcode viewer handles it to figure out how to tie into it.

Kragrathea, 2020-08-20:
Actually I don't think this is a bug anymore. I just tried it and the model showed up even though I didn't accept the size warning. The model just takes a long time to load. 

If you can still reproduce this let me know. Otherwise I will close soon.

closed 2020-08-28 by Kragrathea