Kragrathea/OctoPrint-PrettyGCode #26: Update Loop

jneilliii, 2020-05-02:
You released version v1.0, but the setup.py is still loading v0.3 so results in an update loop. You need to update the line [here](https://github.com/Kragrathea/OctoPrint-PrettyGCode/blob/fdb63153843910037acfba3e0e3c85244250e53b/setup.py#L17) to be `1.01 as well.

MimbaMonkeyHouse, 2020-05-08:
Yup, here too. Do we need to uninstall the old version and reinstall the new version?
Anyone found a solution?

fernandex00, 2020-05-08:
You need to unzip the file, edit the line mentioned in the first comment, create the zip file again, and update the add-on, it should work

Kragrathea, 2020-05-10:
Ok, I think I fixed the issue. Let me know if not.

jneilliii, 2020-05-10:
It did, thanks.

closed 2020-05-10 by jneilliii

gilshahar7, 2020-05-13:
Hey, im sorry for commenting on a closed issue but i am a complete noob, i cant find PrettyGCode in the plugin manager using the OctoPrint interface.. am i missing something?

jneilliii, 2020-05-13:
It's not registered yet in the plugin repository.