Kragrathea/OctoPrint-PrettyGCode #31: Text Color for filename, layers, height, etc lost in background on full screen

msteele999, 2020-06-01:
When I run PrettyGCode in full screen, some dashboard elements are not visible as they are the same color or near the same color as the the background (please see screen shot).

The plugin is absolutely phenomenal and I run all my jobs in this view.

[image]

msteele999, 2020-06-01:
[image]

Kragrathea, 2020-08-18:
Others have reported issues with unreadable text and I am trying to figure out a solution.

Kragrathea, 2020-08-19:
This should be fixed in the next release.

closed 2020-08-19 by Kragrathea