Kragrathea/OctoPrint-PrettyGCode #33: PiCam interference

FHOGPHIL, 2020-08-18:
Installed today and PiCam stopped working  unistalled and it is back

linuxgurugamer, 2020-08-18:
I have PiCAM installed as well, it didn't affect it when I installed.  I have both this and PiCAM working nicely together

FHOGPHIL, 2020-08-18:
Have Octolapse running too. Maybe the three create conflict? 

Please excuse brevity and auto corrects - sent from iPhone!

linuxgurugamer, 2020-08-18:
Octolapse, is that the time-lapse module?  If so, I also had that running.

I have a new install (1 week old), Pi-4, with Octoprint installed, a PiCam v2.  all three worked for me.
What else do you have installed?

FHOGPHIL, 2020-08-18:
Same setup. All with most current updates. 
Did multiple restarts just to confirm. Checked all cables. When could get the video, uninstalled this and everything worked again. 

Please excuse brevity and auto corrects - sent from iPhone!

Kragrathea, 2020-08-18:
Yikes. I don't have a PiCam so I can't check that out. Can you let me know if you are able to narrow it down?

Kragrathea, 2020-08-19:
Did you ever figure this out? I am going to close until there is new info.

closed 2020-08-19 by Kragrathea