Kragrathea/OctoPrint-PrettyGCode #34: Compatibility with 'Floating Navbar'

Naugrimohtar, 2020-08-18:
It seems that there is an issue with this plugin, in full screen mode, with another plugin 'Floating Navbar' where the title bar conflicts with this plugins top buttons.
[image]

Deses, 2020-08-18:
I confirm this issue, it's also happening to me.

Floating navbar buffer is set to 20px, which I think is the default.

Kragrathea, 2020-08-18:
Hmm. The obvious fix would be to move the buttons down. But then then would look funny if you don't have the Floating Nav bar. How about this for a temp fix. Add this CSS to the "Additional CSS" section of the PrettyGCode tab. It will move those four buttons to 40px down from the top.

`.fstoggle{
  top:40px !important;
}
.pgstatetoggle{
  top:40px !important;
}
.pgfilestoggle{
  top:40px !important;
}
.pgsettingstoggle{
  top:40px !important;
}`

Deses, 2020-08-18:
Hey thanks for the that snippet! Based on that I tweaked it a little bit more and here's what I'm using:

`.fstoggle{ top:45px !important; }
.pgstatetoggle{ top:45px !important; }
.pgfilestoggle{ top:45px !important; }
.pgsettingstoggle{ top:45px !important; }
#myslider{ top:7% !important; }
#state_wrapper{ top:80px !important; }
#files_wrapper{ top:80px !important; }`

sr1329, 2020-08-18:
Oh gosh wonderful fix! Now the only issue is the white on white text on the accordion sidebar.

Deses, 2020-08-18:
>Now the only issue is the white on white text on the accordion sidebar.

I don't see any white on white text! 🤔 Care to share a screenshot, please?

sr1329, 2020-08-18:
Yes let me try on my phone. Keep in mind I am using a plug in called themify and it’s default theme which makes all text white and the background dark. It is like a “dark mode” for octoprint.

KenLucke, 2020-08-18:
Confirmed (I was just comig to report this and see the thread already).  Not only does it hide info when FNB is on, but parts of the Dashboard are hidden even when it is not:

ON:
[image]

OFF:
[image]

LarryVand, 2020-08-18:
Sorry if this is in the wrong place but are are all the settings supposed to be saved somehow?? Like the additional CSS code (just discussed) and more importantly to me all the view settings?? Because every time I start Octoprint I have to go into view settings and turn off the Orbit When Idle flag and turn on Sync To Progress... and I apologize if this question has been asked and answered before... Thank you.

Kragrathea, 2020-08-18:
> Sorry if this is in the wrong place but are all the settings supposed to be saved somehow?? Like the additional CSS code (just discussed) and more importantly to me all the view settings?? Because every time I start Octoprint I have to go into view settings and turn off the Orbit When Idle flag and turn on Sync To Progress... and I apologize if this question has been asked and answered before... Thanks.

Those settings are saved in the cookies in the browser. And there are separate settings if you use the direct to FullScreen URL. If it still isn't working start a new Issue on it.

LarryVand, 2020-08-19:
Thanks for the reply. I opened a new issue #43 https://github.com/Kragrathea/OctoPrint-PrettyGCode/issues/43

sr1329, 2020-08-19:
Did a new install on Octopi 18 64 bit. text is fixed and so is dashboard. Wow very quick. Also noticed anti-alias. Really cool.

Kragrathea, 2020-08-19:
I might do a different fix for this down the road. But closing for now.

closed 2020-08-19 by Kragrathea