Kragrathea/OctoPrint-PrettyGCode #36: Feature Request - Themes / Switch GCode Visualizer and Camera

DylanCDuke, 2020-08-18:
Great plugin, good work.

Looking to see if custom themes can be chosen. Looks like it utilizes themes from Themeify. Using discord theme+dashboard plugin, the text is unreadable on the white background.

Curious if this can be set independently to use the default on the plugin, while discord on my octoprint. 

Additionally, would be interesting to switch GCode and Camera, if I'm more interested in the camera view being prominent.

DylanCDuke, 2020-08-18:
Maybe this plugin should just disregard theme? Dashboard has a setting to disregard in fullscreen, if that can be utilized that could fix this issue.

Kragrathea, 2020-08-18:
I don't specifically use Themeify (unless that is what Octoprint uses) so I am not exactly sure what you are asking. You can probably fix the issue with text being unreadable by using the Addtional CSS section on the PrettyGCode tab. What text is unreadable?

DylanCDuke, 2020-08-18:
Hey @Kragrathea - see example here:
https://gyazo.com/340f7d3b2ae63cce1bc8ecad7a2be251
Basically it's using Octoprint theme. So if I set this to default, text is no problem. But if I'm using themify and set to discorded theme, it changes those values on prettygcode plugin.

Honestly it could probably be fixed with custom CSS, I just don't know much CSS

derekslenk, 2020-08-19:
If you figure out the CSS let me know. I am also missing the text

DylanCDuke, 2020-08-19:
Yeah I played around with it a bit and couldn't ever get it to work @derekslenk 

@Kragrathea - might be able to provide fields for a couple elements on the dashboard the users that aren't familiar with CSS like us can just fill in with hex values or something? Just overwrites the default/theme

Kragrathea, 2020-08-19:
The text not being visible in Themes will be fixed in the next release (1.2.1)

DylanCDuke, 2020-08-19:
Thanks Kragrathea! I'll look for this release, and let you know if it's resolved on my end!

Kragrathea, 2020-08-19:
I am going to close this. If it isn't actually fixed file a new issue.

closed 2020-08-19 by Kragrathea

derekslenk, 2020-08-20:
Thank you @Kragrathea !