Kragrathea/OctoPrint-PrettyGCode #41: Bug - PrettyGCode Not able to be reordered

DylanCDuke, 2020-08-18:
Doesn't look like I can reorder using Tab Order, or Dragon Order. Both keep it on the last position

sr1329, 2020-08-19:
Same issues here.

Kragrathea, 2020-08-19:
Hmm. Hey @jneilliii do you have any idea why Dragon Order doesn't work with this plugin? Maybe I have to do something to the tab first?

jneilliii, 2020-08-19:
This might actually be a bug on my side. Recent changes to OctoPrint converted the default GCode viewer into a plugin, and I'm doing a replace on the word gcode with gcodeviewer and it's causing your plugin's ID to get renamed and not match up correctly. It's getting saved as `plugin_prettygcodeviewer` instead of what it should be `plugin_prettygcode`. Let me see if I can fix that real quick for you guys by making the replace more restrictive.

jneilliii, 2020-08-19:
Just release version 0.1.6 of Dragon Order that should fix this.

jneilliii, 2020-08-19:
Just released version 0.5.11 of Tab Order that should fix this.

Kragrathea, 2020-08-19:
Thanks! I am going to close this then.

closed 2020-08-19 by Kragrathea

DylanCDuke, 2020-08-19:
Thanks guys! I'll update when my current print is done and let you know if if this is resolved on my end!