Kragrathea/OctoPrint-PrettyGCode #42: Bug - Nozzle sinks into model when Octolapse move occurs

mneumark, 2020-08-19:
I have Octolapse running which means at some point once per layer it moves the extruder over to left-center (in my case). when this happens in PrettyGCode it just sinks into the model and then pops back when the move is over.

Kragrathea, 2020-08-20:
Apparently Octolapse inserts a GCode command to move the print head to Left Center and that command isn't in exactly the format I expect. If you can figure out exactly what that command is I can try to fix it. Otherwise I will have to come back to this when I can find time to get Octolapse working on my setup. I use my camera for something else.

energiebrand, 2020-08-20:
Yesterday I did my first print with "Pretty GCode". I noticed that a similar behaviour occurs during the last and final layer. The nozzle suddenly droped down to layer 0 and finished there. The gcode was generated by CURA 4.6.1. Unfortunately I did not make a screen shot, but since there a some other GUI problems when using different browsers I will soon open a new issue and add here a screenshot of the dropped nozzle. By the way, I do not use Octolapse!

Kragrathea, 2020-08-28:
Fixed in next release (1.2.2)

closed 2020-08-30 by Kragrathea