Kragrathea/OctoPrint-PrettyGCode #43: [BUG] View settings are not saved using Chrome V 84.0.4147.135

LarryVand, 2020-08-19:
Start Chrome and Octoprint and type 192.168.xx.xx and select the tab PrettyGCode. The default view settings are "Show Mirror" & "Orbit when Idle" ON everything else OFF. Now change the view settings and clear every option except Sync To Progress.  Load an object, zoom on it, etc. make sure everything works.

Now close Chrome.... and restart Chrome... and load Octoprint. Go to PrettyGCode. Click View settings. The settings reverted to the default ON "Show Mirror" & "Orbit when Idle" and everything else off.

I tested both the PrettyGCode tab mode and full view mode. The view settings are NOT saved in either.

Just in case it matters I use the following 3rd party plugins (in alphabetical order): 
ArduCamFocus
Consolidated Tabs
Dashboard
DisplayLayerProgress Plugin
Dragon Order
Floating Navbar
Fullscreen Plugin
Navbar Temperature Plugin
PrettyGCode
PrintTimeGenius Plugin
Simple Emergency Stop
TP-Link Smartplug

Kragrathea, 2020-08-19:
This works as expected for me. Could it be because you have cookies disabled or something?

KenLucke, 2020-08-19:
Would it not be better and more consistent with other plugins to save the settings in config.yaml?  That way it would not matter which browser you used at the moment to open OctoPrint.

LarryVand, 2020-08-19:
I checked everything and cookies are NOT disabled. All other websites work fine and remember the login etc... so I don't know what's going on with PrettyGCode. I'm wondering if any of the other plugins have a role in this.

Kragrathea, 2020-08-19:
The idea is the settings should be per browser so your phone for example doesn't use the same layout and settings as a high end PC. 

I am not sure what is going on with the settings not being saved. If I will see if I can reproduce it again.

LarryVand, 2020-08-19:
I have a question that maybe related... does the Terminal tab of Octoprint that has those 3 settings at the bottom (Suppress temperature messages, etc...) is it supposed to work the same way as PrettyGCode in storing settings?? Cause that isn't work either. Every time I restart Octoprint, at the Terminal tab all the options are cleared and I have select "Suppress temperature messages" again.

BTW, I looked for octo cookies in chrome and these are the ones I found. The 1 in plugins.octoprint.org  is called  cookiesconsent_status. So under what name those PrettyGCode cookies supposed to be stored?
[image]

LarryVand, 2020-08-19:
I just change the Chrome settings to "Allow all cookies" from "Block 3rd party cookies in incognito". Changed the settings. Exited Chrome and restated, and the settings reverted back to default again. 
[image]

Kragrathea, 2020-08-19:
In Chrome you can use the F12 key to bring up the developer tools. In there you can look on the Application tab, Under "Local Storage" there should be a key like the one in the screen shot below. All the options are stored in that value.  

[image]

LarryVand, 2020-08-19:
Now you tell me why in there are just the default?
[image]

LarryVand, 2020-08-19:
Current settings are only Show Mirror. Everything else is unselected. Now you tell me why my view settings do not get used the next time I load Chrome and octoprint...
[image]

LarryVand, 2020-08-19:
BTW, I updated the screenshot and the text, in case you only saw the github emailed of the earlier version....

Kragrathea, 2020-08-19:
At this point I don't know. The settings appear to be saved correctly. And as far as I can tell you are the only one having the issue, so it must be something about your setup. It could be octoprint, browser, network or I don't know what else.  Until I can reproduce it or find another case I don't know what else to suggest.

LarryVand, 2020-08-19:
I'm going to try a different browser (Edge) and see if the problem still happens. Will let you know.

LarryVand, 2020-08-20:
Nope. Edge browser does it also. So it is NOT browser related. I close the Edge browser and when I reopen the settings are back at default. So it must be something to do with either the plugins I have installed or the whole octopi/octoprint configuration... BTW, if you use multiple browsers on the same laptop/desktop/phone/what have you, you need to setup the settings and CSS text all over again.  So here is another vote for config.yaml. It will also fix my problem on top of being able to use different browsers on the same laptop/desktop without having to reenter the settings...

Kragrathea, 2020-08-20:
I sort of reproduced it on Edge. Try using the fullscreen interface URL (like below) and see if the settings are saved. Also try toggling the windows (like camera) to see if that forces a save.   

http://192.168.1.5/?fullscreen=1#tab_plugin_prettygcode

LarryVand, 2020-08-20:
I'll be darn... that worked. And on Chrome at that. So that brings the question, where is the bug at that was causing it not to work before?

Kragrathea, 2020-08-20:
I think it is a race condition based on when the tabs in the UI are initialized. I'll have to dig into it further to figure out how to fix.

Kragrathea, 2020-08-20:
If you use the below style of URL when you first load OctoPrint it should save settings correctly as well. The problem seems to be when you launch on some other tab (say Temperature) and then switch to the PrettyGC tab it doesn't load the settings correctly. Probably a bug/feature in the UI library I am using.  

http://192.168.1.5/#tab_plugin_prettygcode

Kragrathea, 2020-08-29:
Fixed in next release 1.2.2

closed 2020-08-30 by Kragrathea