Kragrathea/OctoPrint-PrettyGCode #46: Intermittent Camera

Shandower, 2020-08-19:
Camera button in full screen not working... usually...

I have two printers each with their own cameras. One is a raspberry pi camera, the other is a USB camera. They both work fine in every other capacity. I've disabled all other camera plugins (except Dashboard), and have tried multiple browsers (safari, Firefox, edge) and have the same results. I updated to OctoPrint-PrettyGCode 1.2.1 this morning, though I saw this in the previous version too.

The camera doesn't display in full screen mode, but if I go into the Dashboard plugin settings and enable the camera, then I will be able to see it in the Dashboard widget on the lower left. If I then reload the page in the browser, the camera will ALSO appear in the lower right as expected. I can then go and disable the Dashboard camera and switch back to PGC full screen and it works fine, until I reload the page.

Dilomos, 2020-08-21:
I can confirm the exact same issue on my system.
and i have also the issue #29 (i have 3d object but no webcam)

Kragrathea, 2020-08-26:
The camera view should be the same one that is visible in the "Control" tab. Does that one work?

The other possibility is that you have two printers. There are some known and unknown issues with that configuration. I am trying to figure out how to test that.

Shandower, 2020-08-26:
Yes, the view in the "control" tab works fine. As I mentioned, it also works fine if I turn on the camera in the Dashboard plugin and view it on the PGC tab in full screen, but it makes the camera part of Dashboard so it's in an awkward spot and semi-transparent.

I only have one printer on each octopi. I have two octopis (completely separate hardware) having the same problem. One is running a Prusa with a USB camera, and the other is running an Ender with a Pi Camera. 

I can make the PGC tab camera work temporarily by turning on the camera in Dashboard, going to PGC full screen, enabling the camera, and then refreshing the web page. I can then go turn off the camera in the Dashboard plugin, and PGC will still show it fine... ...until I refresh that web page, at which point it will go away and not come back.

I've confirmed this on Safari (on Mac), Firefox (Mac and Windows), and Edge (windows). Again, both octopis behave the same.

Kragrathea, 2020-08-26:
Possibly it is conflicting with whatever dashboard is doing with the camera view. Can you try disabling that plugin just long enough to see if if fixes the camera issue. I don't use the embedded camera feature on Dashboard.

Shandower, 2020-08-27:
I disabled Dashboard on both Pis, and on both the PCG Fullscreen camera works as expected.  I re-enabled dashboard, and the previous behavior returned. 

I also attempted to uninstall and reinstall Dashboard, but the problem return with Dashboard again. So at the moment, I can have camera or dashboard, but not both (unless I put the camera IN the Dashboard)

Kragrathea, 2020-08-27:
Good to know. 

Hey @jneilliii. It looks like there is a conflict with how we handle the camera. We are probably trying to do the same thing. Not sure if the fix is on my side or yours or both. I'll look into a fix from my side this weekend.

jneilliii, 2020-08-27:
Dashboard plugin is not mine @Kragrathea, that's @StefanCohen.

Kragrathea, 2020-08-28:
Doh! Sorry. You authored so many plugins I guess sometimes I forget there are some you didn't do. :-)

Kragrathea, 2020-12-05:
Camera issues should be fixed in next release.

closed 2020-12-05 by Kragrathea