Kragrathea/OctoPrint-PrettyGCode #49: Feature Request: Different color for current layer

mneumark, 2020-08-19:
In the normal GCode viewer you can limit it to just the current layer and such.

In 3d like it is in PrettyGCode, its hard to tell where we are in the current layer, it would be cool to show 1 color for printed stuff, 1 color for stuff to be printed on the current layer, and stuff already printed on the current layer. Make all of it optional so people can choose what to show, kinda like in the normal GCode viewer.

Kragrathea, 2020-08-19:
Believe it or not that turns out to be really hard. Mostly because of the tricks I am having to do to make it run fast enough. I basically have to generate all the geometry at load time and that locks in the colors. Only way to change the color is to regenerate the geometry and that is slow and memory intensive.

closed 2020-08-19 by Kragrathea

mneumark, 2020-08-20:
What if you were to put a thin partially transparent rectangle down on the top, and then print on top of that layer, and then move up that sheet when a layer finishes?