Kragrathea/OctoPrint-PrettyGCode #52: Wrong rendering of G2 and G3 gcodes

sameSEA, 2020-08-21:
With G2 and G3 moves picture gets corrupted. 
E.g. G code  processed with ArcWelder in Octoprint:
[image]

TodWulff, 2020-10-23:
Just stopped by to see if someone else had posted this as an issue, as I too am having issues with incorrect depiction by Pretty GCode, of Arc Welder plugin generated g-code (namely G2/G3).  https://i.imgur.com/EoJeRXu.png  If there is anything we can do to help, given you lack a machine that is compatible with these GCodes, I am sure that one of us would be willing to be a bit of a guinea pig.  I hang out at the OctoPrint Discord Server (@PhucKnutt).  Take care.

ChristopheTdn, 2020-10-25:
I add the Arc welder octoprint page. Maybe the explanations can help you.
https://plugins.octoprint.org/plugins/arc_welder/?fbclid=IwAR3JfMrBeRPFaIDxuDC638kTmVe0Y0pCb44HKjgvEtWBooRj-c9WOBMtiI4
Thx for your work !

Shama71, 2020-10-25:
Hi.
Please fix problem.
T
[image]

[image]

Deses, 2020-10-25:
Like the other peeps, I also just installed the newly released ArcWelder and I'm also having this issue.

I'm printing a cylindrical object much like @Shama71 and only the supports or straight lines are visible.

I'm using Marlin 2.0.7.2 with a SKR board on an Ender 3 and arc support is enabled and working ok.

NameOfTheDragon, 2020-10-26:
This suddenly became a lot more important with the release of the Arc Welder plugin. Here's my example:
[image]

FYI, I'm using Klipper firmware on an Ender-3, Klipper supports only the offset (X Y I J) version of the commands and doesn't support the R version. Luckily Arc Welder produces the version that works with Klipper. I've attached teh gcode file that I download from Octoprint after it was processed by Arc Welder.
[Roller_Blind_End_Clip_0.4n_0.2mm_PLA_ENDER3_49m.zip](https://github.com/Kragrathea/OctoPrint-PrettyGCode/files/5440886/Roller_Blind_End_Clip_0.4n_0.2mm_PLA_ENDER3_49m.zip)

LMS0815, 2020-10-27:
> This is going to be a tough one. There are 2 version of each command with various different behaviors. And my printer doesn't support those commands either. I'll have to come back to this issue.

And yo need to check/evaluate the Firmware G90/G91 behavior (See OctoPrint [G90/G91](https://github.com/FormerLurker/ArcWelderPlugin/issues/38) flag).

mriscoc, 2020-11-08:
I have the same problem, model is printed ok, but Gcode visualizer is wrong.
[image]
[image]
[image]

FormerLurker, 2020-11-24:
@Kragrathea, I'd like to help if I can.  Might be able to get arc commands working on your printer too if you give me some info about the make/model and a link to the current firmware.  Even if there are memory issues it might be possible to squeeze out enough to enable arcs.

Oh, nice plugin too!  Sorry I created issues for you :(

Kragrathea, 2020-12-05:
I started on support for this. But I can't use it with my printer due to the fact I don't have enough firmware memory. I have a CR10 and I added a ABL that required me to disable ARC to fit.

Kragrathea, 2020-12-05:
If you want to help implement these GCodes look in PrettyGCode.js for the G2 and G3 cases. And figure out how to make lineSegments out of the arc.

FormerLurker, 2020-12-05:
@Kragrathea, I have already created code to deconstruct arcs into segments, and will take a look to see if it can be applied.  I've got 0 experience with 3d rendering, and haven't examined the code yet, so I can''t promise anything.

FormerLurker, 2020-12-05:
Hmm.. it's a start, lol:

[image]

Looks rough, but it's actually really close.  Will debug and get back to you.

FormerLurker, 2020-12-05:
Getting closer:

[image]

Need to tweak the arc interpolation settings, then I think it should be ready for more testing.  The pic here shows what old marlin with mm_per_arc_setting=1.0 would do (not pretty).  I have added some additional parameters.

Note:  I have not yet implemented the R format, but that won't be too bad.

FormerLurker, 2020-12-05:
Ok, it is looking good now.  I have the beginnings of supporting the R parameter, but until I get some gcode that works it is just commented out with notes.  Here is the obligatory benchy :)

[image]

Kragrathea, 2020-12-14:
Fixed in latest version.

PR #84 (2020-12-06): Add G2/G3 Support
Adds support for G2/G3.  There are a few constants that can be used to adjust the rendering to match Marlin firmware (but not 100% of the options have been added yet).  R and K are not yet supported since I don't know how K works (not in the docs I've seen) and I have no code with which to test R.  There is some code commented out that should be close, but needs to be debugged.  Let me know if anything needs modification, or if you need gcode to test it.  You could always install arc welder and use that to generate arcs in X Y Z I J E F form.

I hope this relieves you of some [... cut]
closed 2020-12-14 by Kragrathea