Kragrathea/OctoPrint-PrettyGCode #57: Feature Request: Sync Layer Without Animation

11chrisadams11, 2020-08-24:
I think it would be nice to have an option to sync the layer progress without having the nozzle animation. On a layer change it would show the current full layer, basically like not syncing to progress and manually moving the slider.

I don't need to see the animation of what is happening, but it's nice to know what the layer is going to look like.

Thanks

Kragrathea, 2020-08-26:
I don't want to add the complexity of another option. But I plan to do something to highlight the current layer at some point and that may be more or less the same thing you want.

Kragrathea, 2020-08-29:
I added an option to disable the nozzle in version 1.2.2. That should be closer to what you want.

11chrisadams11, 2020-08-29:
Yes, that sound like it should work great. Thank you.

closed 2020-08-30 by Kragrathea