Kragrathea/OctoPrint-PrettyGCode #60: Using Backup/Restore  in Octoprint for some reason doesn't Backup/Restore PrettyGCode

LarryVand, 2020-08-28:
Use the backup/restore function of Octoprint V1.4.2.
I excluded uploaded files & metadata from the backup, and I also excluded timelapses from the backup
I then formated a brand new SD card, loaded octopi 0.17.0, updated to V1.4.2 and tried the Backup/Restore function
Every plugin that I had before loaded just fine. All except PrettyGCode. And could very well be that the backup function never backed up PrettuGCode in the first place. Either way PrettyGCode was the only plugin out of 11 that did not restore.

Kragrathea, 2020-08-29:
Sorry, no idea on this one. And it would be a pain to reproduce. Unless someone can point out what I am doing wrong I am not going to try to fix right now.

closed 2020-08-29 by Kragrathea

Kragrathea, 2020-09-11:
This turned out to be a bug in the plugin repository, and It should be fixed soon.

LarryVand, 2020-09-13:
> This turned out to be a bug in the plugin repository, and It should be fixed soon.

That's great! Thanks for following up...