Kragrathea/OctoPrint-PrettyGCode #63: Model not showing in OBS Browser Source even if CPU enabled

zen85, 2020-08-30:
Let me start with how much i love what this plugin provides...

I love to watch the tab in octoprint and everything works perfectly fine there in chrome and other Browsers. However when i want to use it in OBS via Browser Source i dont see the model but i see the printhead with its animations.
After I updated the PrettyGCode today I also dont see the Printhead anymore.

Considering the other open Issues I think this is a related pathing issue?

I also tried to use "http://octopi.local:80/?fullscreen=1#tab_plugin_prettygcode" but its the same issue.

[image]

Kragrathea, 2020-08-30:
If you are even getting the grid then you have the GPU enabled correctly. It looks like it is unable to load the GCode file from OctoPrint. 
-Does it work outside of OBS Studio?
-You mentioned the pathing issue, and that may in fact be the problem, why do you suspect that?

zen85, 2020-08-31:
Yes it works perfectly fine on chrome and edge. 

I think the problem might be somehow related to this issue: #29 
of course i could be wrong.

I also tried with simple and complex models just to see if it has to do with some cache-limitation but that did not seem to make any difference.

Kragrathea, 2020-09-06:
Sorry, I am kind of stuck on this one. I have no idea how to test it since it is only happens in the OBS browser. This is OBS Studio right? And not one of the other streaming programs? I have only used it in OBS Studio.

zen85, 2020-09-17:
dont be sorry. its an overall truly beautiful plugin :)

yes. it is OBS-Studio. I can soon test on another computer with obs-studio and give you an update here...

closed 2020-12-05 by Kragrathea