Kragrathea/OctoPrint-PrettyGCode #66: Camera view does not respect OctoPrint webcam rotation setting

kodek, 2020-09-04:
I just mounted my RasPi camera, and the mount required me to put the camera in a portrait orientation. I corrected this by enabling the OctoPi "Rotate webcam 90 degrees counter clockwise" option.

Unfortunately, the camera view in the PrettyGCode plugin continues to show a landscape view. Refreshing didn't help.

Kragrathea, 2020-09-06:
All PrettyGCode does is visual relocate the web camera from the Octoprint Control Tab.  In my setup the camera is rotated and works correctly. If it is not displaying the same as that view then there must be something messed up in Octoprint or the CSS somewhere.

Kragrathea, 2020-12-05:
I think this might be fixed in next release. I changed how the camera works. If not resubmit.

closed 2020-12-05 by Kragrathea