Kragrathea/OctoPrint-PrettyGCode #67: Infinite G-Code analysis after installation and unable to uninstall

Xyolyp, 2020-09-06:
Since installing this plugin the pre printing G-code analysis takes forever (I canceled it after 15 minutes). 
On top of that the plugin does not appear in the plugin manager as installed and therefore can not be uninstalled.

Kragrathea, 2020-09-06:
PrettyGCode shouldn't do anything at all during Gcode analysis and should be inert unless the tab is displayed. The other problem (if it actually is installed) sounds like something messed up in the Octoprint Plugin Manager. Unfortunately I have no idea how that system works. You will probably have to ask in the Plugin forum.

Xyolyp, 2020-09-07:
Thank you for the hint to get in touch with the plugin manager guys,

closed 2020-09-07 by Xyolyp