Kragrathea/OctoPrint-PrettyGCode #68: Plugin ID is not consistent with that of the repo, causes issues

cp2004, 2020-09-11:
Plugin ID is not consistent with that of the repo, causes issues (a) migrating to Python 3 and (b) restoring from a backup.

Plugin id is `prettygcode` - the plugin *repository* is `PrettyGCode`.

@Kragrathea - Is it OK if I change the id on the repo? To be honest, you haven't really got a choice since the ID *should never be changed* within the plugin, as it breaks updates. 🙂

jneilliii, 2020-09-11:
@Kragrathea I can update the plugin repo for the correct id in lieu of changing it on your end.

jneilliii, 2020-09-11:
So this was strange @cp2004 it seems the json file is built from the filename and not the supplied id in the jinja template, because it was correct in his md file, but the md file was named PrettyGCode. Very confusing, but I just renamed it and the id is coming in now all lowercase.

cp2004, 2020-09-11:
Spotted the commit come through on Discord... bug in the plugin repository?

jneilliii, 2020-09-11:
I assume it's a bug in one of the build actions.

closed 2020-09-11 by cp2004

Kragrathea, 2020-09-11:
Hmm. Not totally understand this. :-) Is there anything I can do on my side to help clean it up?

cp2004, 2020-09-11:
@Kragrathea do not worry, it is now sorted. The plugin file was named with capitals in the repository, different from the plugin ID in `setup.py` (`PrettyGcode` vs `prettygcode`)

Discovered when (a) restoring a backup or (b) upgrading to Python 3.

jneilliii, 2020-09-11:
Just be aware that if you ever need to make changes to your plugin listing on the official repo, you will need to make sure to pull the current version first with the proper non-capital naming.