Kragrathea/OctoPrint-PrettyGCode #73: Visualizing the position of item on grid

Tinker0000, 2020-10-22:
I'm having a weird issue where the pretty item is showing up at the extreme lower right hand corner. Is there any way to place the item in the center of the grid while printing? When I initially open the PrettyGCode tab in the interface, the grid pattern is centered (in the viewing area) but there is no item. I have to zoom the picture way out to locate the item, then use the mouse to get the item in the center of the viewing area. The standard GCodeViewer in OctoPrint does the same thing except the item appears in the upper right hand corner. 
[image] 
[image] 
[image] 
[image]

zoombahh, 2020-10-25:
what printer do you have? does it have 0,0 at the center of the bed instead of on one corner?

Tinker0000, 2020-10-27:
I am using a Creality CR-10S4. I have the "Origin at Center" box checked in OctoPrint, should I try unchecking the box and see what happens?

zoombahh, 2020-10-27:
does the printer go to the center of the Bed when you command it to 0,0,0 ?

unchecking origin at center will probably fix your visual issues, but the setting should of course reflect the reality of your printer.

on my cr-10 the 0,0,0 origin is at the front left corner, so i have origin at center unchecked.

Tinker0000, 2020-10-27:
Thanks for your quick response, I'll take a closer look at my settings and try unchecking the box.

Tinker0000, 2020-10-31:
I changed the origin to the left corner instead of center in my printer profile (These are the only two options). The Pretty G visualizer still shows as printing on the extreme lower corner, I guess I'll just have to learn to live with it in both tabs. It really isn't that big of a deal. Thanks for your suggestions.

Kragrathea, 2020-12-05:
I am not sure I support Origin at center correctly. It could also be that your bed size is set too wide in octoprint.

Tinker0000, 2020-12-06:
Thanks for your reply, I just learned to deal with it. I will close the
case as unresolved.

Tinker0000, 2020-12-06:
Funny thing, I just looked at the OctoPrint App Goode viewer and it does
the same thing. See picture attached.

Kragrathea, 2020-12-06:
I suspect your bed dimensions are not set properly in octoprint.