Kragrathea/OctoPrint-PrettyGCode #78: Issue with rendering

CaptainSweatpants, 2020-11-01:
Noticed this started happening recently. The only change to my octoprint was adding ArcWelder plugin (not sure if that's causing this) but as you can see perimeter walls are not being rendered, travel lines are showing, and other top/bottom lines are either missing or out of wack. 

[image]

t413, 2020-11-04:
Same issue here. PrettyGCode doesn't support G2/G3 _arc move_ commands, just the standard G0/G1 linear moves. The built-in _GCode Viewer_ plugin supports arcs, interestingly. 

Took a look at the code but it's pretty terse javascripty javascript.. not my specialty.

jpjokela, 2020-11-06:
[image]
Yep, definitely Arc-Welder. Seems to draw lines between arc segments though (or something)

NameOfTheDragon, 2020-11-13:
Duplicates Iss #52

Kragrathea, 2020-12-05:
This is because arc is not supported yet. See issue #52

closed 2020-12-05 by Kragrathea