Kragrathea/OctoPrint-PrettyGCode #80: "soft" buttons not displaying info tabs.

sonjamichelle, 2020-11-26:
For the past week or so I've been having trouble with the Pretty GCode plugin.

The "soft" buttons that display the info don't do anything. the only button that works it the webcam toggle.

I did have ArcWelder installed, tried uninstalling it, but no change.

I am running
OctoPrint version : 1.4.2
OctoPi version    : 0.17.0
on a Raspberry Pi 3B+

I have tried Chrome, Firefox and Edge on about 5 different computers.

I even tried rebuilding the image, no change.

This is all I get:

[image]

Kragrathea, 2020-12-05:
I am not sure what is going on here. I released a new version. Can you try that and see if you still have the issue?

sonjamichelle, 2020-12-07:
I installed the new version. No  change.  Though when I hit the browser back button, the panels show up. This happens in ALL browsers on ALL my PC's on ALL operating systems. We're talking Chrome, Firefox, Edge and Windows, Linux, Android and PC, R-Pi, VMWare. Short of uninstalling every plugin and starting from scratch to see if there is a conflict somewhere I'm not sure what to do. I've already uninstalled all my Chrome extensions, no change after that. Browser code has gotten so muddled these days I don't who is using what code or if they are all using the same "codebase" *shrug*

Could there be something in my router settings? I do have a pretty restrictive setup blocking ads, filth, and other stuff. my hosts file is rather extensive.

It was just one day everything was working, then it wasn't. I have since rebuilt my OctoPi server a few times since. Stands to reason that it should have corrected itself. So it has to lie somewhere in my PC, Browser or Router settings.

Really Really puzzling.  I'm am by no means a coder so I am completely stumped, though I am by no means uninitiated when it comes to PC's and Tech. (Did IT for over a decade back in the day.)

Any out of the box ideas?

Kragrathea, 2020-12-07:
I am not totally sure what you mean by "soft" buttons. Does the button that opens the settings window work? Does the fullscreen button?

All the buttons for files and stuff do is toggle visibility of the element in CSS. If the Files element shows up in the regular OctoPrint interface it should show up in the PGC interface. 

Only thing I can think of is to open the browser developer interface (F12 on windows) and see if there are any messages in the console that help.

sonjamichelle, 2020-12-07:
The only ones that work are the ones on the right side of the screen. The camera toggle, settings and full screen toggle. The ones on the left do nothing.

I have to admit my HTML is pretty dated. I included a screenshot of the main PrettyGCode page to see if it helps.

https://www.designcreations.us/images/pretty_gcode_screenshot.png

Kragrathea, 2020-12-07:
That might help. Click the Console tab on the right side (next to Elements) and then reload the tab and click the buttons that don't work. See if any errors show up in the console log.

Kragrathea, 2020-12-07:
As for the root cause. If it does fail on all browsers then it is most likely a conflict with some other octoprint plugin. You should try disabling anything that isn't part of the stock install and see if that helps.

sonjamichelle, 2020-12-07:
Here is what I get with the console:

https://www.designcreations.us/images/pg_1.png

https://www.designcreations.us/images/pg_2.png

As far as the next step, I have a print in progress, so when that completes I can try killing off other plug-ins and see who may be the culprit.

Kragrathea, 2020-12-07:
There are quite a few errors in there. But as far as I can tell none of them are specific to PGC.  Try disabling other plugins and let me know.

sonjamichelle, 2020-12-11:
Sorry it has taken me so long to respond and report, been busy on the homefront.

I went and disabled EVERY 3rd party plugin save for PG, rebooted the system, and PG worked as expected.

I then began enabling my plugins one by one. When I got to Consolidated Tabs (https://github.com/jneilliii/OctoPrint-ConsolidatedTabs) PG began to ac up again. I disabled that Consolidated Tab, rebooted and PG worked as expected.

I then went through, enabled the rest of my 3rd party plugins save for Consolidated Tabs, in fact I completely removed it and PG continued to work as expected.

So it look like Consolidated Tabs is the culprit.

I also opened an issue on Consolidated Tabs GitHub

closed 2020-12-14 by Kragrathea