Kragrathea/OctoPrint-PrettyGCode #86: Z sometimes wrong when simulating Arcs

Kragrathea, 2020-12-08:
When simulating a print the animation for Arcs sometimes uses the wrong Z height.

FormerLurker, 2020-12-08:
I will see if I can recreate this with simple gcode, and will post it once that is complete.

Kragrathea, 2020-12-08:
I took a quick look but couldn't immediately see the problem.

As far as debugging it goes. 
=Z shouldn't change during the course of an arc. It should only change with at new layer. And if Z isn't provided as part of the G command it should be assumed to be the same as the last Z. 
-You could try checking the Z value against parserLastExtrudedZ which I use to detect layer changes.

FormerLurker, 2020-12-08:
I suspect the way I extract z is having problems.  It's different than the model builder code which has methods for ensuring the coordinates are absolute.  If I had to guess what's wrong, that's where I would look.

FormerLurker, 2020-12-09:
So, this may be solved.  I don't see the old behavior anymore.  The issue boiled down to values being reported as NaN, but that wasn't handled.

I thing some adjustments could be made to the animation routine still, but will have to see how it works on a live print.

Kragrathea, 2020-12-09:
Cool! Now that you mention it, I think I had the same issue. I see I added IsNan() around everything. I wish I could remember _why_ the values were NaN. Do you happen to know?

FormerLurker, 2020-12-09:
I do not.  I didn't look into it much, and noticed you were catching those as well.  I did create a function you could use so that they all use the same method (check for undefined and nan).  I could look into this friday though.  I'd like to see if I can figure out the animation bit.

Kragrathea, 2020-12-10:
This fix works for the Z issue. 

I am having performance issues, but I don't think it is related to this. It might be related to the buffer overflowing. I am looking into that now and will make a separate issue if needed.

closed 2020-12-10 by Kragrathea

FormerLurker, 2020-12-10:
I had some large memory usage, but chalked it up to viewing a 150mb file..  will take a look and comment in any new issues I see pop up.

Kragrathea, 2020-12-10:
The way the adaptive buffer works is it speeds up the printhead the longer the buffer is. To overflow you would have to add hundreds of segments at once. Which may be actually be the correct behavior and I just need to make the buffer bigger.

`rate=rate*(buffer.length/5.0);`

FormerLurker, 2020-12-10:
Maybe an self growing buffer is the way to go?  you could create a function to add to the buffer, and if it hits max_len, grow it by 2x:

```
self.addToBuffer = function(x) {
     if (buffer.length == max_length)
     {
          max_length = max_length * 2;
          buffer.length = max_length * 2;
     }
     buffer.push(x);
}
```

I literally just typed that in, so some debugging/syntax adjustments might be required.

Kragrathea, 2020-12-10:
This isn't a buffer we want to grow. If it overflows the print head animation just glitches a bit. 

I am going to start a new issue for the buffer overflow.

commit ff7919bc (2020-12-09): Handle NaN in coordinates for arc simulation.  May resolve #86
commit ccd36060 (2020-12-14): Test branch (#92)

* Add Arc Support (G2/G3) for X Y Z I J E F form.

* Add initial print simulation support for G2/G3.

* Remove unused code block

* Add gauge

* Handle NaN in coordinates for arc simulation.  May resolve #86

* Handle NaN for i,j,r,e and f.

* Add status bar for temps

* Update status text

* Show status in fullscreen only

* Update status message

* Update version