Kragrathea/OctoPrint-PrettyGCode #88: Printhead simulation buffer overflows occasionally.

Kragrathea, 2020-12-10:
It maybe that we need just a bigger buffer. But I want to make sure the problem isn't something else. For example, too many segments in an arc when building the 3d model could cause perf issues.

FormerLurker, 2020-12-10:
@Kragrathea, take a look at those arc interpolation parameters.  In particular, the min_mm_per_arc_segment may need to be increased.  This needs to be lowish while printing, but might not need to be so low for visualization.  Perhaps something like .25 or 0.5 would work.  I'd try the lower of the two first, since zooming in will definitely reveal the fine grained detail, but just doubling it to 0.2 would cut the segments in half.

FormerLurker, 2020-12-10:
I have to clarify.  I said doubling that value would cut the segments in half, but that is a worst case, and would only apply to arcs of a small radius.  Even still, it depends on the other constants.  Been working too long today, and I'm a bit foggy.  Will take a closer look on Friday.

Kragrathea, 2020-12-10:
At least in my case the buffer overflow was caused by the tab loosing focus for a while. That disables the render thread so the 
 the printhead stops updating and eventually the buffer overflows. I'll see if I can come up with a better way to handle it or disable the console warning if not.

Kragrathea, 2020-12-10:
As far as number of segments generated. I just tested the new arc commands on a Benchy. Before Arc conversion it had 335k lines and after it has 268k. Now obviously that is good all around for rending. :-) But does it follow that it should have that many fewer? It doesn't point to some other issue does it?

FormerLurker, 2020-12-10:
Are you talking about before and after conversion, or are you talking about the number of segments after interpolation?  I have seen 90+% reduction in line count at the extreme end after arc welding, but typically the number of segments after interpolation is greater than or equal to the number of segments in the original file.  It depends on firmware settings, of course.

Kragrathea, 2020-12-10:
I counted the number of line segments added to the 3d model before and after. 

I would expect that there would be more segments after. But I guess it would depend very much on the settings used when interpolating the arc when it generates the 3d model.

FormerLurker, 2020-12-10:
Unless you are usiny a specific version of cura that produced too many points, i too would be concerned.  If you can post the before gcode I will run it through arc welder and arc straightener (the inverse processor) and see what the results are.  It might lead to a reevaluation of the visualization interpolation.

Kragrathea, 2020-12-10:
Before and after. I am about to check in a change to TestBranch that (among other things) prints the number of lines to the console after the object loads.

[BenchyTest.zip](https://github.com/Kragrathea/OctoPrint-PrettyGCode/files/5669807/BenchyTest.zip)