Kragrathea/OctoPrint-PrettyGCode #93: Displayed amount of layers incorrect

fluefiske, 2020-12-15:
Hey guys,

Love your plugin very much, I discovered though that the amount of layer displayed is incorrect.

[image]

The real amount should be 897.

Keep up the good work, love it,

André

Kragrathea, 2020-12-15:
I think that is showing 64013 layers. My guess is your slicer is generating code that has a lot of Z changes in it for some reason?

rholak, 2021-02-21:
@fluefiske Was this a spiral mode print? If so, that will cause this behavior since the Z changes are almost constant instead of every 0.2mm (or whatever resolution you choose).  I am printing a spiral vase right now and the plugin is showing over 40,000 layers.

fluefiske, 2021-02-21:
Yes it was