Kragrathea/OctoPrint-PrettyGCode #95: Rendering is shown only in tiny window in lower-left corner

timur-tabi, 2020-12-20:
This is with Octoprint 1.5.2 running on a Mac with OS X 10.11.6

[image]

Kragrathea, 2020-12-21:
Does it render correctly in the tab mode?

timur-tabi, 2020-12-21:
It does the same thing in tab mode.  Just a small rectangle in the lower-left.

Kragrathea, 2020-12-22:
The only thing I can think of is a plugin conflict. What other ones do you have installed? Maybe try disabling them to see if can figure out if that is the problem.

timur-tabi, 2020-12-24:
Looks like it's a browser bug.  It only happens on Firefox on one of my Macs.  With any other browser or any other computer (including other Macs), it works fine.  I even tried switching to safe mode, but that only made it worse (no image at all).

closed 2020-12-24 by timur-tabi