Kragrathea/OctoPrint-PrettyGCode #121: Nozzle - Incorrect height

testbughub, 2021-08-22:
Hi!

I just installed the plugin and started a print, when I noticed that the nozzle is going around at the wrong height:  
[image]

I looked around all the issues, but I couldn't find anything on this problem, altough I found #100, but I'm not sure if it's related.

If I just missed some information, I apologize.

Kragrathea, 2021-08-22:
When you first refresh a page it can take a minute or so to update the nozzle to the correct height. This is due to the way the octoprint sends gcode to the printer. I can only tell the Z height when the layer changes. If you wait does the nozzle eventual start using the correct height?

testbughub, 2021-08-23:
Yep, you were right. I gave it a couple of minutes and it started working fine.
Thanks for the quick reply, and the plugin, of course!

closed 2021-08-23 by testbughub