Kragrathea/OctoPrint-PrettyGCode #126: Rotate view to match camera?

WilfulTravesty, 2021-09-15:
Is there any way to rotate the view to better match the location of my camera? I can zoom in and out, but the view always sees the print from the front right while my camera is in the front left. It would make it easier to following if they were closer to the same position, especially when I pop up the camera view over the PrettyGCode view.

Kragrathea, 2021-09-16:
There will be a new version of the plugin soon that allows you to save the camera position.

WilfulTravesty, 2021-09-24:
Is there currently a way to change the camera position in the view? (Such that it will save that position in the new version?) I didn't see one when I went looking.

Kragrathea, 2021-09-24:
The new version has an option to "Save Camera". Turn that on and move the camera wherever you want it. Then when you reload the window the camera will be in that position.

The new version isn't on the plugin update yet since it is still in testing. If you want to try it see here:

https://github.com/Kragrathea/OctoPrint-PrettyGCode/discussions/128

melyux, 2025-04-25:
"soon" ™️