Kragrathea/OctoPrint-PrettyGCode #129: Crash with MacOS Safari v15.0

cbradburne, 2021-09-24:
Crashes and restarts OctoPrint when opening PrettyGCode while using latest version of Safari (v15.0) on MacOS 10.15 and 11.6.

Kragrathea, 2021-09-24:
Hmm. Is that a NEW version of Safari or something? This version of the plugin has been out a long time and I haven't had any other reports of crashing.

cbradburne, 2021-09-26:
Hi, yes it's the latest version of Safari on MacOS 10.15 and 11.6. It was working fine on previous versions and still works fine when I use Chrome. This is happening on 2 different Macs. Obviously worth waiting on more reports in case it's just an issue on my end.

My current Safari version is, Version 15.0 (16612.1.29.41.4, 16612).

pandel, 2021-09-29:
+1
As soon as I click on the PrettyGCode tab the page starts reloading with a "This webpage is reloaded because a problem occurred" message. Sadly, Javascrip console or developer tools do not help in finding the issue...

Kragrathea, 2021-09-29:
This is going to be tricky to find since I don't have a Mac. 

- Can you confirm that this used to work before the new version of Safarii?
- So I am clear, the message "This webpage is reloaded because a problem occurred" is a Safari error message and not an OctoPrint error correct?

pandel, 2021-09-30:
@Kragrathea 
Yes, it worked before and yes, the message is a Safari error message.

sfarkhan, 2021-09-30:
I can confirm this as well. Two Octoprint instances, both crash consistently on the new Safari 15.0

Kragrathea, 2021-09-30:
Searching online I found lots of, probably, worthless advice about clearing caches and cookies.  Could someone please try this and see if it makes any difference?

https://discussions.apple.com/thread/8316218

Beyond that it seems there may be larger issues with Safari 15, some people claim banking sites now show the same error. We may have to wait for a Apple patch Safari again. :-(

sfarkhan, 2021-09-30:
I just tested in in Safari with cleared cache/cookies and it consistently crashes the tab and reloads on both instances. Sorry :-/  FWIW, it works in Chrome Version 94.0.4606.61 (Official Build) (x86_64) and Firefox 92.0.1 (64-bit). Let me know if I can help in any other way.

JWRu, 2021-10-02:
I have the same problem since I updated to Safari 15.

Kragrathea, 2021-10-09:
Does the log give any hints as to what the problem is?

pandel, 2021-10-10:
> Does the log give any hints as to what the problem is? 

Sorry, not at home atm, can tell you more tomorrow.

pandel, 2021-10-11:
@Kragrathea Don't know, if you understand, what is happening here, but FWIW:

```Process:               com.apple.WebKit.WebContent [63187]
Path:                  /Library/Apple/*/WebKit.framework/Versions/A/XPCServices/com.apple.WebKit.WebContent.xpc/Contents/MacOS/com.apple.WebKit.WebContent
Identifier:            com.apple.WebKit.WebContent
Version:               16612 (16612.1.29.41.4)
Build Info:            WebKit-7612001029041004~8 (612A78a)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Safari [58544]
User ID:               501

Date/Time:             2021-10-08 13:00:50.219 +0200
OS Version:            macOS 11.6 (20G165)
Report Version:        12
Bridge OS Version:     5.5 (18P4759a)
Anonymous UUID:        1240831F-7616-CD66-FEC4-CAC79C4A818A

Sleep/Wake UUID:       77E3F442-BEB3-4D26-9AA1-75486AAAAE32

Time Awake Since Boot: 33000 seconds
Time Since Wake:       11000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [63187]

VM Regions Near 0:
--> 
    __TEXT                      10ae47000-10ae4b000    [   16K] r-x/r-x SM=COW  /Library/Apple/*/WebKit.framework/Versions/A/XPCServices/com.apple.WebKit.WebContent.xpc/Contents/MacOS/com.apple.WebKit.WebContent

Application Specific Information:
Bundle controller class:
BrowserBundleController
 

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.apple.WebCore             	0x000000010d05993c WebCore::GraphicsContextGLOpenGL::reshapeDisplayBufferBacking() + 188
1   com.apple.WebCore             	0x000000010c65f243 WebCore::GraphicsContextGLOpenGL::reshapeFBOs(WebCore::IntSize const&) + 451
2   com.apple.WebCore             	0x000000010c6612c3 WebCore::GraphicsContextGLOpenGL::reshape(int, int) + 547
3   com.apple.WebCore             	0x000000010da4ffe4 WebCore::WebGLRenderingContextBase::initializeNewContext() + 2324
4   com.apple.WebCore             	0x000000010da5b312 WebCore::WebGLRenderingContextBase::create(WebCore::CanvasBase&, WebCore::GraphicsContextGLAttributes&, WebCore::GraphicsContextGLWebGLVersion) + 2882
5   com.apple.WebCore             	0x000000010d91525a WebCore::HTMLCanvasElement::createContextWebGL(WebCore::GraphicsContextGLWebGLVersion, WebCore::GraphicsContextGLAttributes&&) + 330
6   com.apple.WebCore             	0x000000010d914e92 WebCore::HTMLCanvasElement::getContext(JSC::JSGlobalObject&, WTF::String const&, WTF::Vector<JSC::Strong<JSC::Unknown, (JSC::ShouldStrongDestructorGrabLock)0>, 0ul, WTF::CrashOnOverflow, 16ul, WTF::FastMalloc>&&) + 1186
7   com.apple.WebCore             	0x000000010c9c2b6b WebCore::jsHTMLCanvasElementPrototypeFunction_getContext(JSC::JSGlobalObject*, JSC::CallFrame*) + 411
8   ???                           	0x00005b7d232011d8 0 + 100593018343896
9   com.apple.JavaScriptCore      	0x0000000110823c4e llint_entry + 112071
10  com.apple.JavaScriptCore      	0x0000000110824939 llint_entry + 115378
11  com.apple.JavaScriptCore      	0x0000000110823c4e llint_entry + 112071
12  ???                           	0x00005b7d238673ae 0 + 100593025053614
13  ???                           	0x00005b7d2375c476 0 + 100593023960182
14  ???                           	0x00005b7d238a7811 0 + 100593025316881
15  com.apple.JavaScriptCore      	0x0000000110823bc9 llint_entry + 111938
16  com.apple.JavaScriptCore      	0x0000000110823bc9 llint_entry + 111938
17  com.apple.JavaScriptCore      	0x0000000110823c4e llint_entry + 112071
18  com.apple.JavaScriptCore      	0x0000000110823c4e llint_entry + 112071
19  ???                           	0x00005b7d2345fbb7 0 + 100593020828599
20  ???                           	0x00005b7d2344bf17 0 + 100593020747543
21  ???                           	0x00005b7d23462509 0 + 100593020839177
22  ???                           	0x00005b7d2381e437 0 + 100593024754743
23  com.apple.JavaScriptCore      	0x0000000110823c4e llint_entry + 112071
24  com.apple.JavaScriptCore      	0x0000000110823c4e llint_entry + 112071
[... 891 lines cut ...]
MALLOC                           136.6M       82 
MALLOC guard page                   32K        6 
Memory Tag 240                       8K        2 
Memory Tag 241                     296K        4 
STACK GUARD                       56.1M       19 
Stack                             17.1M       20 
VM_ALLOCATE                        912K       10 
WebKit Malloc                    303.9M      218 
__CTF                               759        1 
__DATA                            44.6M      633 
__DATA_CONST                      34.0M      416 
__DATA_DIRTY                      2368K      292 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       544.4M       43 
__OBJC_RO                         70.3M        1 
__OBJC_RW                         2496K        2 
__TEXT                           538.4M      622 
__UNICODE                          588K        1 
mapped file                       83.9M       93 
shared memory                      144K        5 
===========                     =======  ======= 
TOTAL                             98.9G     2674 
TOTAL, minus reserved VM space    35.2G     2674 
```

Kragrathea, 2021-10-11:
@pandel That does help a little. It is apparently crashing when it inits WebGL. Looking online it seems other WebGL apps are also having issues. 

There is one thing you can try. Someone with a different app has fixed the crash by changing some Safari settings.  In "Settings -> Safari -> Advanced -> Experimental Features" -- there should be a bunch of "GPU Process" Experimental WebKit Features. Try turning off (or on) "Canvas Rendering" and/or "GPU Process: WebGL"

pandel, 2021-10-11:
@Kragrathea 
WOW! Many thanks! I had to **enable** "GPU Process: WebGL" and restart Safari. Now PrettyGCode works and WebKit does not crash anymore. Kudos!

BTW, Safari 15 has no experimental option "Canvas Rendering" ;-)

Kragrathea, 2021-10-11:
Nice! I will make a note in the Readme to **enable "GPU Process: WebGL" and restart Safari.** 

The other option should have been "GPU Process: Canvas Rendering". But it doesn't matter now. 

@cbradburne @sfarkhan @JWRu @pandel

pandel, 2021-10-11:
@Kragrathea Ah, I see! This other option was already enabled, btw.