Kragrathea/OctoPrint-PrettyGCode #140: Nozzle always at Z level 0

Xolverr, 2022-07-25:
Screenshot from iPhone XR Chrome, but also occurs on PC and Mac Chrome. Not a terrible issue just not pretty 😂
[image]

Kragrathea, 2022-08-26:
Not sure what would cause that. And I haven't had any other reports of this. Maybe something about the slicer you are using?

Xolverr, 2022-08-26:
I use Cura. Most of the time after a little while it does fix itself, so it might have something to do with an unstable http connection. (I was using mobile data in the screenshot)

Kragrathea, 2022-08-28:
Oh, There is a known issue where the Z height is only updated when the layer changes. This is because the Gcode leaves out the Z part until it changes. I seem to remember it was tricky to fix that bug for some reason...

Xolverr, 2022-08-28:
Awesome, that sounds like its probably the problem, the model I was printing at that time was decently large. Thanks!

closed 2022-08-28 by Xolverr