Kragrathea/OctoPrint-PrettyGCode #143: Webcam stopped working in PGC after ubuntu 22.04 upgrade

Claghorn, 2022-10-29:
[https://community.octoprint.org/t/webcam-not-working-in-prettygcode-after-ubuntu-22-04-upgrade/47404](url) has details. The webcam view worked fine when my x86 ubuntu box was on 20.04, but after upgrade to 22.04 I fixed several things (like the syntax in the haproxy config file), but clicking on the webcam button in PrettyGcode full screen mode does nothing. (I did get new versions of OctoPrint and PrettyGcode after the upgrade, so it wasn't just ubuntu I upgraded). Octoprint is currently 1.8.6 and PrettyGcode 1.2.4.

In the control tab and the settings for testing webcam and timelapse, everything works fine. I can even talk directly to the mjpg streamer locally and from another system and see the webcam, it just doesn't work in PGC.

isaacolsen94, 2023-02-16:
This happens for me as well. Did you ever figure out how to fix it?

Claghorn, 2023-02-16:
I haven't had the patience to try and dig through the maze to figure anything out. I'm considering reinstalling everything starting with ubuntu and building octoprint from scratch to see if that works and it was just some little glitch in the update I never found. But if it still doesn't work after that, it would be a huge waste of time :-).

Claghorn, 2023-02-16:
This inspired me to poke around some, and I'm guessing it has something to do with the new version of haproxy which I had to guess how to configure since the old config wouldn't work. The instructions I followed long ago had an haproxy entry recognizing things named webcam and I see the source code for the PCG plugin appending strings like "/webcam/?action=stream" to things it looks up. None of the other screens that show the webcam use the /webcam uri syntax, so while I was able to get haproxy to stop complaining about my config, I probably didn't get it to do the right thing.

isaacolsen94, 2023-02-16:
Oh interesting, I'll have to poke around as well.

As for the first post about building from scratch. My setup is brand new (as of 4 days ago :P) and I haven't been able to get it to work. So I don't think that would help unfortunately

Claghorn, 2023-02-16:
Once I gave up trying to get haproxy to tell me exactly what it was doing and switched to wireshark to see the full http requests, I could see that it isn't transforming the uri as intended. Not knowing anything about what the gibberish in haproxy.cfg actually means, I'm not sure how to fix it. This chunk of config is probably what's wrong:

`backend webcam`
`        http-request replace-uri ^([^\ :]*)\ /webcam/(.*)     \1\ /\2`
`        option forwardfor`
`        server webcam1  127.0.0.1:8080`

That should be directing the uri to port 8080 (which it does), and getting rid of the /webcam/ part of the uri (which it doesn't do). So if I could only figure out the magic gibberish I need to change, it might work.

Claghorn, 2023-02-16:
Success at last! The magic gibberish involved using a different command:

`backend webcam`
`        http-request replace-path /webcam/(.*)     /\1`
`        option forwardfor`
`        server webcam1  127.0.0.1:8080`

It finally works in pretty gcode!

isaacolsen94, 2023-02-20:
> he uri as inten

Awesome!! Nice job figuring this out! I'm going to give it a shot now.