Kragrathea/OctoPrint-PrettyGCode #144: Can not access top buttons in fullscreen mode

dchotwolf101, 2023-01-26:
[image]

The buttons are under the header bar and are hard to click.

FindusD, 2023-02-05:
I have the same problem, in full screen mode the buttons at the top and at the bottom are hidden under the header and footer from OctoPrint.
To see the buttons at the bottom I can scroll down, but I cannot do anything to see the button at the top.

Guenni75, 2023-04-29:
Hi.
Same Problem here.
I think it's a problem with the UI Customizer Plugin.

gregbarbosa, 2023-04-30:
I'm also using UI Customizer, this CSS pushes down the buttons alongside hiding the status bar (that for some reason only ever shows as 0 for me).

1. Open OctoPrint Settings
2. Under Settings > Plugins > UI Customizer > Advanced > Custom CSS, paste in the following:

```css
.pgfullscreen #tab_plugin_prettygcode .gwin {
    margin-top: 40px;
}

.pgfullscreen .pgstatus {
    display: none;
}
```

Before
[image]

After
[image]

TBFMcDuck, 2024-02-11:
If you want to get rid of the navbar completely you can use this CSS instead (from [this issue](https://github.com/Kragrathea/OctoPrint-PrettyGCode/issues/112)

`.pgfullscreen #navbar {
visibility:hidden;
}
`