Kragrathea/OctoPrint-PrettyGCode #160: After upgrading Octoprint to version 1.10.0 PrettyGCode shows empty window

anderl1969, 2024-04-28:
After upgrading Octoprint to 1.10.0 the PGcode-plugin most of the time doesn't work:

- it shows an empty window
- no reaction to the buttons in the top right corner (toggle settings window, toggle fullscreen mode)

Sometimes the plugin works as usual, but most of the time it doesn't.

Rylan-Meilutis, 2024-05-01:
It appears to be an issue with the way the js gets the volume of the bed, here is the error in the chrome logs

```
packed_core.js?8d71fd9c:1828 Error calling onTabChange on view model PrettyGCodeViewModel : Cannot read properties of undefined (reading 'volume')
TypeError: Cannot read properties of undefined (reading 'volume')
    at updateBedVolume (https://octoprint.rylanswebsite.com/static/webassets/packed_plugins.js?47991bcf:52262:75)
    at PrettyGCodeViewModel.self.onTabChange (https://octoprint.rylanswebsite.com/static/webassets/packed_plugins.js?47991bcf:52046:29)
    at callViewModelIf (https://octoprint.rylanswebsite.com/static/webassets/packed_core.js?8d71fd9c:1827:75)
    at https://octoprint.rylanswebsite.com/static/webassets/packed_core.js?8d71fd9c:1821:138
    at Pn (https://octoprint.rylanswebsite.com/static/webassets/packed_libs.js?409a9b49:30:530)
    at Function.<anonymous> (https://octoprint.rylanswebsite.com/static/webassets/packed_libs.js?409a9b49:51:66)
    at callViewModelsIf (https://octoprint.rylanswebsite.com/static/webassets/packed_core.js?8d71fd9c:1821:95)
    at callViewModels (https://octoprint.rylanswebsite.com/static/webassets/packed_core.js?8d71fd9c:1820:56)
    at OctoPrint.coreui.exports.onTabChange (https://octoprint.rylanswebsite.com/static/webassets/packed_core.js?8d71fd9c:1854:47)
    at HTMLAnchorElement.<anonymous> (https://octoprint.rylanswebsite.com/static/webassets/packed_core.js?8d71fd9c:1877:1896)
```

cansinacarer, 2024-05-06:
I have the same issue.

Rylan-Meilutis, 2024-05-06:
Right now I am in school for about the next 3 weeks, if school lets out and there has been no reply I will fork this repo and try to resolve the issue.

gitolicious, 2024-05-10:
What I found: Navigating to the start page http://octoprint:5000/ (or any other tab), then clicking on the PrettyGCode button, everything works fine. Opening the tab directly with http://octoprint:5000/#tab_plugin_prettygcode leads to the above error because the `self.printerProfiles` are not populated.

jacopotediosi, 2024-06-28:
The culprit here is that the printer profiles are asynchronous and it is not certain that they have already been populated when the PrettyGCode tab is opened. When the tab is opened, this plugin tries to call the `updateBedVolume()` function, which tries to access the printer profiles and crashes because they are not yet populated.

In fact, as also noted by @gitolicious, if the tab is opened when the OctoPrint web page is already completely loaded, the printer profiles are already populated and this bug does not occur, while if the page is loaded directly on the tab, they are not yet ready:
> Navigating to the start page http://octoprint:5000/ (or any other tab), then clicking on the PrettyGCode button, everything works fine. Opening the tab directly with http://octoprint:5000/#tab_plugin_prettygcode leads to the above error because the `self.printerProfiles` are not populated.

The OctoPrint default temperature graph was having (almost) the same issue in OctoPrint 1.10.0rc3: https://github.com/OctoPrint/OctoPrint/issues/4980#issuecomment-2031357981

My PR #162 should fix this.

rdenis1, 2024-06-28:
> The culprit here is that the printer profiles are asynchronous and it is not certain that they have already been populated when the PrettyGCode tab is opened. When the tab is opened, this plugin tries to call the `updateBedVolume()` function, which tries to access the printer profiles and crashes because they are not yet populated.
> 
> In fact, as also noted by @gitolicious, if the tab is opened when the OctoPrint web page is already completely loaded, the printer profiles are already populated and this bug does not occur, while if the page is loaded directly on the tab, they are not yet ready:
> 
> > Navigating to the start page http://octoprint:5000/ (or any other tab), then clicking on the PrettyGCode button, everything works fine. Opening the tab directly with http://octoprint:5000/#tab_plugin_prettygcode leads to the above error because the `self.printerProfiles` are not populated.
> 
> The OctoPrint default temperature graph was having (almost) the same issue in OctoPrint 1.10.0rc3: [OctoPrint/OctoPrint#4980 (comment)](https://github.com/OctoPrint/OctoPrint/issues/4980#issuecomment-2031357981)
> 
> My PR #163 should fix this. In the meantime, you should be able to try my patched plugin version by installing the following URL via the OctoPrint plugin manager: https://github.com/jacopotediosi/OctoPrint-PrettyGCode/archive/patch-1.zip

Your fix works great!

KevWal, 2024-07-29:
Just a +1 your fix works for me too, thank you!

darkvertex, 2024-07-29:
I too confirm the fix works. Thank you! 😄

PR #163 (2025-04-06): Fix crash printerProfiles not yet populated
Fix #158, fix #160
commit f4d6223e (2025-04-06): Fix crash printerProfiles not yet populated (#163)

Fix #158, fix #160
closed 2025-04-06 by Kragrathea